sblantipodi/arduino_bootstrapper #5: AP page shown on mobile is different from docu

Reneg973, 2021-04-11:
Mobile gives a lot more entries like IP. What to input there for DHCP?

sblantipodi, 2021-05-09:
Hi, Arduino Bootstrapper does not support DHCP currently.
It's a design choice and a question of tastes.
I will probably add an option to let you choose a fixed IP or use DHCP in the future.

Thanks for your interest. Closing the issue but feel free to reply if you want.

closed 2021-05-09 by sblantipodi

Reneg973, 2021-05-09:
It was not a question for a new feature. The code is already there if I see it correctly. Your already implemented a wifi client. So if I first enter the AP server IP I am able to put in an existing SSID. So I thought that after rebooting it will choose this SSID to connect to and get an IP. But maybe I was wrong and it always creates an AP with the given IP?