sblantipodi/arduino_bootstrapper #7: PingESP utility issue

Pronoe, 2021-12-23:
Hello,
Thank you for your bootstrapper library.
I am using it to build my project of pool filtering pump management based on an ESP32.
I use also some pieces of code from your Smartostat.
I found an issue with PingESP utility because it does not seem to be ESP32 ready.
After including ESP32 definition, the compiler goes on but I still have an issue with the "extern "C" {#include <ping.h>}" declaration.
As I understand, this call for a pre-compiled ping C code.
Thank you for your help.
Patrick

Pronoe, 2021-12-23:
Hello,
I have solved the issue by installing the library ESP32Ping from https://github.com/marian-craciunescu/ESP32Ping and including the corresponding header in PingESP.h.
Its now compiling but the code is too big for the target (1.5 Mo for 1.3 Mo max) despite the operational code is limited to basic functions up to now.

sblantipodi, 2021-12-30:
Hi,
thank you for opening the issue Pronoe.

ESP8266 have built in functions to ping other devices while ESP32 don't have this features builtin.
That utility in particular works for ESP8266 only, if you want to use it on ESP32 you need an external lib.

There is no other way on ESP32 unfortunantly.
Feel free to reopen the issue if you need it :)

Pronoe, 2021-12-31:
Hi Davide,

 

Thank you for your reply.

I did use an external library as mentioned in the issue.

I suggest however that you modify the PingESP.h file as proposed in the attached file.

 

Meanwhile I have learnt that the “external C” directive was to link embedded functions of the micro controller.

Regards

Patrick

Pronoe, 2022-03-08:
After migration of my application to Bootstrapper 1.12.6 release, I have encountered some issues with PingESP because I had merged some modifications that I made locally for the ESP32 and your last modifications for the ESP8266. As I had also installed ESP8266 library for checking, I think the compiler might have merged some specific code for the ESP8266 and code for the ESP32.
Since then, I got in trouble with the WiFi link and my gateway was putting the link down for all the connected device.
It seems that I have finally solve this issue by adding a `PingClass pingESP` declaration in my own header file and keeping 
```
#elif defined(ESP32) 
    #include <WiFi.h>
    #include <ESP32Ping.h>
    #include <ping.h>
``` 
in the PingESP.h header file.
I will propose later on a pull request for this file and to the README.md file to provide additional guidelines for ESP32 users.

sblantipodi, 2022-03-09:
thank you!!! I'll reopen it until your pull request...