sblantipodi/arduino_bootstrapper #8: Suspected bug with writing setup file with ESP32

Pronoe, 2021-12-25:
Hellow,
in WiFiManager.cpp, when writing the setup file, in the case of ESP32, the current code is:
File configFile = SPIFFS.open("/setup.json", "w");
            if (!configFile)
            {
              Serial.println("Failed to open [setup.json] file for writing");
            }
            serializeJsonPretty(doc, Serial);
            serializeJson(doc, configFile);
            configFile.close();
            Serial.println("[setup.json] written correctly");
The else block that exists for the ESP8296 is missing.
The code should be:
File configFile = SPIFFS.open("/setup.json", "w");
            if (!configFile)
            {
              Serial.println("Failed to open [setup.json] file for writing");
            }
            else
            {
              serializeJsonPretty(doc, Serial);
              serializeJson(doc, configFile);
              configFile.close();
              Serial.println("[setup.json] written correctly");

Is this correct ?
The same apply for BootstrapManager::writeToSPIFFS

sblantipodi, 2021-12-30:
I'm sorry can you elaborate what is the suspected bug? 
Thank you.

Pronoe, 2021-12-31:
Hi Davide,

 

In line #528 of the original file “WifiManager.cpp”, you are testing the correct opening of the file : 

if (!configFile) {

Serial.println("Failed to open [setup.json] file for writing");

}

But there is no else block after this, the lines starting from #531 are executed even if the file has not been successfully opened :

serializeJsonPretty(doc, Serial);

       serializeJson(doc, configFile);

       configFile.close();

       Serial.println("[setup.json] written correctly");

 

You even have a message stating that the file has been written correctly, when in fact the opening might have failed.

 

However I recognised that this is not a serious issue because I think there is no real risk  that the file could not be opened.

 

A similar issue applies for BootstrapManager::writeToSPIFFS, starting from line #323 in the original file “BootstrapManager.cpp”

An “else” condition is missing, then write operation are performed event if the file failed to open.

This is more annoying because an opening error can occur if the file name is wrong for example.

 

Regards

Patrick

sblantipodi, 2022-01-06:
Thank you for reporting Pronoe, fixing it soon.

commit 432029e7 (2022-01-07): closes #8
closed 2022-01-07 by sblantipodi