sblantipodi/arduino_bootstrapper #10: DNS configuration issue with ESP32 platform

Pronoe, 2021-12-30:
Hello,
in the WifiManager::setupWiFi routine, there is a call to WiFi.config(microcontrollerIP, IP_gateway, IP_DNS) for both ESP8266 and ESP32.
However, in the time.h library associated to the ESP32, the correct parameters are :
WiFi.config(microcontrollerIP, IP_gateway, **Subnet_mask**, IP_DNS1, IP_DNS2), with IP_DNS1 & 2 optionnal.
This result to the DNS IP being interpreted as a subnet mask value.
Associated with the use of a static IP for the microcontroller, this prevents access with domain name such as NTP servers for example.
After implementing the correct parameters list, all is working fine.
I think the time.h library is slightly different for the ESP8266 but I cannot check this, because this platform is not installed on my system.
Regards.
Patrick

sblantipodi, 2021-12-30:
Thank you for reporting this issue, I'll fix it soon :)

Pronoe, 2021-12-31:
Hi Davide,

thank you for your reply.

Patrick

Pronoe, 2022-03-02:
Hello David,
it seems that you missed to fix this issue in release v1.12.6.
Your code in WifiManager.cpp  is nearly unchanged:
```
 if (!microcontrollerIP.equals("DHCP")) {
    WiFi.config(IPAddress(helper.getValue(microcontrollerIP, '.', 0).toInt(),
                          helper.getValue(microcontrollerIP, '.', 1).toInt(),
                          helper.getValue(microcontrollerIP, '.', 2).toInt(),
                          helper.getValue(microcontrollerIP, '.', 3).toInt()),
                IPAddress(helper.getValue(IP_GATEWAY, '.', 0).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 1).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 2).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 3).toInt()),
                IPAddress(helper.getValue(IP_SUBNET, '.', 0).toInt(),
                          helper.getValue(IP_SUBNET, '.', 1).toInt(),
                          helper.getValue(IP_SUBNET, '.', 2).toInt(),
                          helper.getValue(IP_SUBNET, '.', 3).toInt()));
    Serial.println(F("Using static IP address"));
    dhcpInUse = false;
  }
```
Only `dhcpInUse = false;` has been added.
My code for solving the ESP32 issue is the following:
```
 if (!microcontrollerIP.equals("DHCP"))
  {
#if defined(ESP8266)
    WiFi.config(IPAddress(helper.getValue(microcontrollerIP, '.', 0).toInt(),
                          helper.getValue(microcontrollerIP, '.', 1).toInt(),
                          helper.getValue(microcontrollerIP, '.', 2).toInt(),
                          helper.getValue(microcontrollerIP, '.', 3).toInt()),
                IPAddress(helper.getValue(IP_GATEWAY, '.', 0).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 1).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 2).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 3).toInt()),
                IPAddress(helper.getValue(IP_DNS, '.', 0).toInt(),
                          helper.getValue(IP_DNS, '.', 1).toInt(),
                          helper.getValue(IP_DNS, '.', 2).toInt(),
                          helper.getValue(IP_DNS, '.', 3).toInt())); // initial code
#elif defined(ESP32)
    WiFi.config(IPAddress(helper.getValue(microcontrollerIP, '.', 0).toInt(),
                          helper.getValue(microcontrollerIP, '.', 1).toInt(),
                          helper.getValue(microcontrollerIP, '.', 2).toInt(),
                          helper.getValue(microcontrollerIP, '.', 3).toInt()),
                IPAddress(helper.getValue(IP_GATEWAY, '.', 0).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 1).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 2).toInt(),
                          helper.getValue(IP_GATEWAY, '.', 3).toInt()),
                IPAddress(helper.getValue(IP_SUBNET, '.', 0).toInt(),
                          helper.getValue(IP_SUBNET, '.', 1).toInt(),
                          helper.getValue(IP_SUBNET, '.', 2).toInt(),
                          helper.getValue(IP_SUBNET, '.', 3).toInt()),
                IPAddress(helper.getValue(IP_DNS, '.', 0).toInt(),
                          helper.getValue(IP_DNS, '.', 1).toInt(),
                          helper.getValue(IP_DNS, '.', 2).toInt(),
                          helper.getValue(IP_DNS, '.', 3).toInt())); // code added by Pronoe on 31/12/2021
#endif
    Serial.println(F("Using static IP address"));
    dhcpInUse = false;
  }
```
and in Configuration.h
```
// IP_SUBNET
#ifndef SUBNET_IP
#define SUBNET_IP "255.255.255.0" 
#endif
const char* const IP_SUBNET = SUBNET_IP;
```
I have downloaded the v1.12.6 release and is testing it with my pump management application.
I will propose later on, merge requests for this current issue and two other opened issues.

sblantipodi, 2022-03-02:
you're right @Pronoe I missed that part of the DNS.
thank you for pointing it out.

pull request (aka merge request) will be really appreciated. 👍

sblantipodi, 2022-03-02:
but is it possible to use a "default DNS" for people who don't want to configure the DNS?

sblantipodi, 2022-03-03:
I reopened the issue since we need some more work here :)

Pronoe, 2022-03-03:
After checking the code of the WiFi library for ESP32 (in WIFISTA.cpp), it appears that when you are using a static IP, it is set through a call to WiFi.config and you have to provide at the same time a DNS1 and DNS2 IP.
However, there is a default value that is 0.0.0.0 and is correct if you don't use DNS access.
With this default value, the controller will not be able to access external address through domain name.
This is not an issue if you dont need at all external internet access or if you want to access to an external fixed IP.
In my own case, I use NTP server and I prefer to use a domain name such as pool.ntp.org because their IP address may vary.

There is no default value other than 0.0.0.0  because a valid default value would be the one provided by your local router, then this depends on your specific router. Or it could be an external DNS provider such as Google (IP 8.8.8.8) but the library may not impose such an external DNS server.
There is a solution described in [https://techtutorialsx.com/2020/04/21/esp32-arduino-set-static-ip-address/](url).
It consists in connecting first in DHCP mode, then retrieve gateway IP and DNS IP provided by the router. Then you connect with a fixed IP and provide the gateway IP and DNS IP retrieved at the first step.
Or you can simply read these data in the router configuration GUI and use them in the Configuration.h file of boostrapper.
I think that boostrapper users are experienced people who will easily find the right solution if they want to use a fixed IP and domain names
We could add some comments in the Configuration file to provide this guiding information.

Pronoe, 2022-03-03:
Hello,
I have installed the ESP8266 library to check the WiFi library code for this platform.
The release 1.12.6 has also bugs for the ESP8266 because the parameters list for `wifi.config` is: `ip, gateway, subnet, dns1`.
The DNS IP is then missing.
My son who is far more experienced than me in software engineering, has instructed me how to manage changes with git.
I will make a pull request very soon to fix this issue.
Two files will be modified: WifiManager.cpp and Configuration.h

sblantipodi, 2022-03-06:
thank you very much Pronoe 👍

commit a47a3114 (2022-01-07): closes #10
PR #17 (2022-03-16): Fix dns issue esp32
- fix DNS issue with ESP32
- remove remaining unnecessary delays
- add printing of connecting attempts information (useful with poor WiFi link)
Tested on ESP32 platform with the use of static IP and DNS Google server.
Closes #10
commit 578c9c6d (2022-03-16): Fix dns issue esp32 (#17)

* fix DNS IP issue and remove unnecessary delays
* fix typo
#Closes #10
closed 2022-03-16 by sblantipodi