sblantipodi/arduino_bootstrapper #11: How to set specific constant overriding those of the Configuration.h file

Pronoe, 2021-12-31:
Let's take as an example, the gateway IP.
In the Configuration.h file of the library, one can find:
    // GATEWAY IP
    #ifndef GATEWAY_IP
    #define GATEWAY_IP "192.168.1.1"
    #endif
    const char* const IP_GATEWAY = GATEWAY_IP;
As I don't want to modify the original library files, I have included my own file MyConfiguration.h which contains:
    #define GATEWAY_IP "192.168.0.254"
Thanks to the #ifndef GATEWAY_IP directive, I expected that my own definition would be taken into account.
But this is not the case, probably due to the fact that the library Bootsrapper is compiled before my sketch.
So at the moment, I had to modify the original file inside the library folder.
What do you suggest to solve for this problem ?

sblantipodi, 2022-01-06:
Hi @Pronoe 
that option can be changed by using build flags.

if you are using platformio you can set it in your platformio.ini file, example here:
https://github.com/sblantipodi/arduino_bootstrapper/blob/d063cb5f03f33a1ba72fc32f69ced05d318088ff/platformio.ini#L27

I don't use Arduino IDE anymore but I'm pretty sure that you can do it without problems in that IDE too.
this could give you an idea on how to do it with Arduino IDE.
https://forum.arduino.cc/t/how-to-set-compiler-constants-required-for-a-library/578762

Pronoe, 2022-01-07:
Hi @sblantipodi 
thank you very much for your answer and your help.
I have tried recently to use platformio but it has appeared a bit complex for me who is a newbie. In addition this raised some conflicts with the Arduino IDE add-on under VScode. As I had some on-going projects with Arduino IDE, I have uninstalled platformio. I will try again later on.
I will review the link that you provided for Arduino IDE, may be this the solution for now.

closed 2022-01-07 by Pronoe

sblantipodi, 2022-01-08:
@Pronoe thank you very much for all your ideas and for the help you gived me in the issues you opened, 
the closed ones has been merged in the main branch, it will take a bit for the release, I would like to close all the issues before releasing...
thank you!!!