sblantipodi/arduino_bootstrapper #23: Examples needs an ino file

iplayfast, 2022-03-13:
The examples folder has a cpp and .h file, but there is no ino file.

Pronoe, 2022-03-14:
You just have to rename `ChangeName.cpp` into `ChangeName.ino` and place it and the file `ChangeName.h`  into a folder named ChangeName so that you can edit and compile them with the Arduino IDE.
You will need also to edit the `Configuration.h` file of the library to adapt it to your own network configuration.

Have a glance also to the [Smart Thermostat](https://github.com/sblantipodi/smart_thermostat) application from [sblantipodi](https://github.com/sblantipodi/arduino_bootstrapper/commits?author=sblantipodi) to better understand all the potential of the library and the way to use it.

sblantipodi, 2022-03-16:
thanks @Pronoe for helping here... 
Closing the issue since Pronoe has answered it already :)

closed 2022-03-16 by sblantipodi